TJREVERB/pfs #53: Investigate Problems with ThreadHandler

abagali1, 2019-11-08:
# Description
On the 11/7/19 meeting, @sparthaje encountered a bug while unit testing `aprs` with `pty` ports. In his  `aprs` class he initialized a `process` meant to continuously read from a serial port and recognize if anything has been received from the serial port. When he initialized a `ThreadHandler` object his `listen()` process terminated unexpectedly, but when he initialized a `Thread` object with the exact same arguments, the process reacted properly. This has led us to suspect that `ThreadHandler` has an internal bug that is hindering processes. @AkashBhave is the original developer of `ThreadHandler`, @sparthaje encountered the bug.

Assigning @AkashBhave to debug the existing `ThreadHandler`, @sparthaje to assist with debugging, or to assist @abagali1 in rewriting a `ThreadHandler` class.

If the bug is found, open a new issue with the `bug` label and open a new branch to fix the issue. In addition to opening a new issue to track the development process, submit a comment on this(current) issue detailing what the bug was and the proposed fix

If the bug cannot be found and a rewrite of `ThreadHandler` is necessary, rewrite `ThreadHandler` with functionality proposed in Issue #49(Handling RuntimeErrors in processes)

AkashBhave, 2019-11-08:
Don't see why a rewrite is necessary, but this is something to look into.

abagali1, 2019-11-08:
Yeah I don’t want this to take too much time so a rewrite is an absolute worst case scenario

sparthaje, 2019-11-14:
ThreadHandler automatically marked all new Theads as daemonized Threads so they were all terminated as soon as the main loop terminated

closed 2019-11-14 by sparthaje