TJREVERB/pfs #69: Investigate the use of pypy in flight

abagali1, 2019-11-19:
# Description
`pypy` is a type of Just In Time(JIT) interpreter that compiles commonly used python files for faster use afterward. For example, in Cycle 2, `submodules` begin to import `core` to pass into `ThreadHandler`. This causes `core` to be imported in each `submodule` file, which reruns the `core` code on every import. With `pypy`, the `core` class would be compiled once, and the compiled version would be accessed later on instead of rerunning the code every time.

# Questions
* Does `pypy` support Python 3.6+
* Does `pypy` support the ARM CPU architecture
  * specifically the Raspberry Pi Zero
* How does using `pypy` change the way `pFS` is run
* Does the benefit outweigh the cost?

abagali1, 2019-11-19:
Link to pypy [here](pypy.org)

cigBot, 2019-12-05:
How is the progress coming with pypy integration?

closed 2020-01-30 by abagali1