TJREVERB/pfs #71: Bug: APRS echo and serial port exceptions

abagali1, 2019-12-06:
On send, APRS receives the same sent message. Investigate if this is groundstation or fixable or (worst case) code for that case

This exception also occurred
```
Traceback (most recent call last):
  File "/home/pi/pfs/helpers/threadhandler.py", line 55, in run
    self.target()
  File "/home/pi/pfs/submodules/radios/aprs.py", line 118, in listen
    result = self.serial.read()
  File "/home/pi/.local/lib/python3.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

```

abagali1, 2019-12-14:
How was this resolved?

sparthaje, 2019-12-14:
I used a try catch block whenever serial.read or serial.write was used, and then handled the error in the aprs class

closed 2020-01-18 by sparthaje